Dilbert66/esphome-components #3: Bot blocks for too much time

fra87, 2024-10-16:
Hello

I'm getting these warnings in the logs:

```
[17:12:19][W][component:237]: Component telegram_bot took a long time for an operation (369 ms).
[17:12:19][W][component:238]: Components should block for at most 30 ms.
```

Measuring the `WebNotify::loop()` function, I see that all this time is spent in function `mg_mgr_poll`.

Is this to be expected?

Dilbert66, 2024-10-16:
Yes, it's to be expected. It does not cause problems.  ESPhome added this annoying warning about long running tasks.

fra87, 2024-10-18:
Ok; I guess there is nothing to do then (e.g. splitting some function or using callbacks) ;) Thanks

closed 2024-10-18 by fra87

Dilbert66, 2024-10-18:
The mongoose lib uses all callbacks for most functions.  The issue with this particular delay is during the connection and SSL setup to telegram which occurs on message send and on reconnect for the long poll. When no messages are sent or received, there is no delay.   The long poll is facilitated with a callback instead of a blocking loop.    Please note though that the moogoose lib is a third party lib not of my creation.  I don't have an easy fix to use a callback for the actual connection delay itself other then using a function on a secondary core task (dual core esp32).  I have played around with trying to get that work right but was initially not impressed with result so did not pursue it.   I do use core tasks successfully for my vista library for the time sensitive functions.