Dilbert66/esphome-components #5: Connecting to API and other debug messages

fra87, 2024-10-19:
I'm almost sure this is fine, but I'm asking just to be 100% sure

In my logs I see

```
[...]
[10:14:44][W][component:237]: Component telegram_bot took a long time for an operation (368 ms).
[10:14:44][W][component:238]: Components should block for at most 30 ms.
[10:14:59][D][telegram:367]: Free memory:  3168
[10:15:19][D][telegram:367]: Free memory:  3168
[10:15:34][D][telegram:360]: Connecting to api
[10:15:34][D][telegram:278]: TLS init - Before: freeheap: 204296,minheap: 157336,maxfree:143360
[10:15:35][D][telegram:283]: TLS init - After: freeheap: 176416,minheap: 157336,maxfree:143360
[10:15:35][W][component:237]: Component telegram_bot took a long time for an operation (373 ms).
[10:15:35][W][component:238]: Components should block for at most 30 ms.
[10:15:39][D][telegram:367]: Free memory:  3168
[10:15:59][D][telegram:367]: Free memory:  3168
[10:16:05][D][telegram:360]: Connecting to api
[10:16:05][D][telegram:278]: TLS init - Before: freeheap: 204300,minheap: 157336,maxfree:143360
[10:16:05][D][telegram:283]: TLS init - After: freeheap: 175808,minheap: 157336,maxfree:143360
[...]
```

And this goes on.

We already spoke about the warnings ("took a long time", #3), but are the other debug messages also expected?

From what I understood:

- The periodic "Free memory" is a feature; every 20 seconds the available free memory is pushed by the loop function, so this is expected
- The "Connecting to api" and "TLS init" are linked to a connection attempt, which happens when `network::is_connected()` is `true` and `connected` is `false`

I'm more interested in the second part, since it seems to me that this happens too "frequently" so I'm wondering if I just forgot something (such as to correctly set a keepalive or something like that), since the connection to the server drops every  30-60 seconds (variable)

Dilbert66, 2024-10-20:
I use the "long poll" method for checking for messages.  This entails opening an SSL connection to the telegram api which will then stay open for a maximum of 120seconds waiting for new messages.  This is using a callback so is non-blocking.  When a message arrives, the callback is called for receiving the message.  Now, what will also happen is that this connection can timeout before the max of 120 seconds for various normal reasons. Either way, when this happens, the previous connection is dropped and a new connection is opened and the wait cycle goes on.   If an outbound message needs to go out, the pending inbound poll is closed , a new tx connection is opened, the message is sent, then the long poll is re-opened again.  The longest part of this process is the initial SSL init.    So what you are seeing is normal when you have polling enabled.  The free memory and tls init messages are just debug prints I put in as I was optimizing for minimal heap useage.  They are not indicative of any issues. I just wanted to be aware of how much heap is used by the SSL session and variables.

fra87, 2024-10-20:
Ok so the messages are ok. As I said, I was "almost" sure, but I wanted to get rid of that almost ;)

closed 2024-10-20 by fra87