Dilbert66/esphome-components #6: Error

Allkorolev, 2024-10-25:
Greetings, I'm trying to run your component in my project, but an error occurs during compilation:

`In file included from src/esphome/components/telegram_bot/telegram_bot.cpp:1:
src/esphome/components/telegram_bot/telegram_bot.h:161:69: error: expected template-name before '<' token
 template<typename... Ts> class TelegramPublishAction : public Action<Ts...> {
                                                                     ^
src/esphome/components/telegram_bot/telegram_bot.h:161:69: error: expected '{' before '<' token
src/esphome/components/telegram_bot/telegram_bot.h:216:76: error: expected template-name before '<' token
 template<typename... Ts> class TelegramAnswerCallBackAction : public Action<Ts...> {
                                                                            ^
src/esphome/components/telegram_bot/telegram_bot.h:216:76: error: expected '{' before '<' token
src/esphome/components/telegram_bot/telegram_bot.h:237:75: error: expected template-name before '<' token
 template<typename... Ts> class TelegramDeleteMessageAction : public Action<Ts...> {
                                                                           ^
src/esphome/components/telegram_bot/telegram_bot.h:237:75: error: expected '{' before '<' token
src/esphome/components/telegram_bot/telegram_bot.h:256:73: error: expected template-name before '<' token
 template<typename... Ts> class TelegramEditMessageAction : public Action<Ts...> {
                                                                         ^
src/esphome/components/telegram_bot/telegram_bot.h:256:73: error: expected '{' before '<' token
src/esphome/components/telegram_bot/telegram_bot.h:291:77: error: expected template-name before '<' token
 template<typename... Ts> class TelegramEditReplyMarkupAction : public Action<Ts...> {
                                                                             ^
src/esphome/components/telegram_bot/telegram_bot.h:291:77: error: expected '{' before '<' token
src/esphome/components/telegram_bot/telegram_bot.h:320:46: error: expected template-name before '<' token
 class TelegramMessageTrigger : public Trigger<RemoteData> {
                                              ^
src/esphome/components/telegram_bot/telegram_bot.h:320:46: error: expected '{' before '<' token
src/esphome/components/telegram_bot/telegram_bot.h:320:46: error: expected unqualified-id before '<' token
*** [.pioenvs/weather-station/src/esphome/components/telegram_bot/telegram_bot.cpp.o] Error 1`

The code that I use even in minimal execution is as follows:
`
external_components:
  - source: github://Dilbert66/esphome-components@dev #uncomment to use github repository version
  #- source:  my_components #uncomment to use alocal directory  called "my_components"
    components: [telegram_bot,mg_lib] #components to load
    refresh: 10min  #frequency to check for new versions

telegram_bot:
  id: webnotify
  bot_id: "123467890:SAMPLETELEGRAMBOTKEYSAMPLETELEGRAM"
  chat_id: "4535645671"
  bot_enable: true
  allowed_chat_ids:
    - "4535645671"

  on_message:
    - text: "*"
      then:
        - telegram.publish:
            message: "Received: ${x.text}"
            to: "${x.chat_id}"`

Accordingly, I replaced all the bot and chat IDs

Dilbert66, 2024-10-25:
Ok, this is fixed on dev now. Missing esphome.h include

Allkorolev, 2024-10-25:
Thanks, good job.
Tell me, is it possible to interact with a bot only through a group? Interaction by writing directly to him is not provided for?

Dilbert66, 2024-10-25:
You can chat with the bot directly using a user id also. Set the chat_id accordingly.

Allkorolev, 2024-10-25:
OK, and the last question is, is it possible to implement dynamic bot_id and chat_id so that you can change them in the web interface without updating the firmware?

Dilbert66, 2024-10-25:
You can create your own api actions to set whatever values you want using lambda actions.  These are the telegram calls you can make:
```
  webnotify->set_bot_id(std::string bot_id) ;
  webnotify->set_chat_id(std::string chat_id) ;
  webnotify->add_chatid(std::stringchat_id) ;
  webnotify->set_bot_enable(bool enable);
  webnotify->set_send_enable(bool enable) ;
```

closed 2024-12-17 by Allkorolev