Dilbert66/esphome-components #7: "in expansion of macro 'IPADDR_NONE'  #define INADDR_NONE   IPADDR_NONE"

vvzvlad, 2024-11-07:
this config is built with an error

```
esphome:
  name: lock-controller
  friendly_name: lock_controller

esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  ssid: "LTE-----"
  password: "-----"
  ap:
    ssid: "Lock-Controller Fallback Hotspot"
    password: "8vDV19WUTQPV"

api:
  encryption:
    key: "----="

ota:
  - platform: esphome
    password: "----"

external_components:
  - source: github://Dilbert66/esphome-components@dev #uncomment to use github repository version
    components: [telegram_bot,mg_lib] #components to load
    refresh: 10min  #frequency to check for new versions

mg_lib:   

telegram_bot:
  id: webnotify
  bot_id: "------"
  chat_id: "------" #default chat id to send to. Can also be a group id.
  bot_enable: true #if you dont need to have the receive bot set to false. Only publish action will be available.
  allowed_chat_ids:  #chat id's that are allowed to send cmds. Can also be group id's. Use "*" to allow all.
    - "-4574033653"
```

```
INFO ESPHome 2024.10.2
INFO Reading configuration /config/lock-controller.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing lock-controller (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Removing unused dependencies...
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/lock-controller/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/esp32/core.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/esp32/gpio.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/esp32/preferences.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/esphome/ota/ota_esphome.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/json/json_util.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/md5/md5.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/mdns/mdns_esp32.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/mdns/mdns_host.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/mdns/mdns_libretiny.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/mg_lib/mongoose.c.o
Compiling .pioenvs/lock-controller/src/esphome/components/network/util.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/ota/ota_backend.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/ota/ota_backend_arduino_libretiny.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/ota/ota_backend_arduino_rp2040.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/safe_mode/safe_mode.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/socket/lwip_sockets_impl.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/socket/socket.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/telegram_bot/telegram_bot.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/wifi/wifi_component.cpp.o
Compiling .pioenvs/lock-controller/src/esphome/components/wifi/wifi_component_esp32_arduino.cpp.o
In file included from /config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/ip_addr.h:43,
                 from /config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/inet.h:45,
                 from /config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/netdb.h:42,
                 from /config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/port/esp32/include/netdb.h:33,
                 from src/mongoose.h:111,
                 from src/esphome/components/telegram_bot/telegram_bot.h:4,
                 from src/esphome/components/telegram_bot/telegram_bot.cpp:1:
/config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/ip4_addr.h:63:37: error: expected ')' before numeric constant
 #define IPADDR_NONE         ((u32_t)0xffffffffUL)
                             ~       ^~~~~~~~~~~~
/config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/inet.h:71:29: note: in expansion of macro 'IPADDR_NONE'
 #define INADDR_NONE         IPADDR_NONE
                             ^~~~~~~~~~~
/config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/IPAddress.h:95:18: note: in expansion of macro 'INADDR_NONE'
 extern IPAddress INADDR_NONE;
                  ^~~~~~~~~~~
*** [.pioenvs/lock-controller/src/esphome/components/telegram_bot/telegram_bot.cpp.o] Error 1
========================= [FAILED] Took 15.10 seconds =========================
```

Dilbert66, 2024-11-07:
This should be working now. New fix pushed to "dev" branch.

vvzvlad, 2024-11-07:
> This should be working now. New fix pushed to "dev" branch.

New error:
```
Compiling .pioenvs/lock-controller/lib195/libsodium/crypto_hash/sha512/hash_sha512.c.o
Compiling .pioenvs/lock-controller/lib195/libsodium/crypto_onetimeauth/crypto_onetimeauth.c.o
Compiling .pioenvs/lock-controller/lib195/libsodium/crypto_onetimeauth/poly1305/donna/poly1305_donna.c.o
/config/lock-controller.yaml: In lambda function:
/config/lock-controller.yaml:67:45: error: 'using element_type = class esphome::http_request::HttpContainer' {aka 'class esphome::http_request::HttpContainer'} has no member named 'body'
                   std::string response_text = response->body;
                                             ^~~~
Compiling .pioenvs/lock-controller/lib195/libsodium/crypto_onetimeauth/poly1305/onetimeauth_poly1305.c.o
Compiling .pioenvs/lock-controller/lib195/libsodium/crypto_pwhash/argon2/argon2-core.c.o
Compiling .pioenvs/lock-controller/lib195/libsodium/crypto_pwhash/argon2/argon2-encoding.c.o
*** [.pioenvs/lock-controller/src/main.cpp.o] Error 1
```

Dilbert66, 2024-11-07:
No idea what that is.  The telegram code does not use the http code.  What is in your yaml at line 67?  It looks to be some custom lambda you have in there that is causing the issue.

vvzvlad, 2024-11-07:
Oops, I think it's my mistake, there was some extra code in there. With the latest dev fix the example is built correctly, yes.

closed 2024-11-08 by Dilbert66