Dilbert66/esphome-components #8: Updating from /dev branch of Dilbert66/esphome-dsckeybus to component based install breaks arm-away

jescarri, 2024-11-17:
First of all, thank you so much for making this extension, I Recently moved from the old way of installing the pluging (Downloading the early days source code to homeassistant etc ) to this external component way, I'm having troubles setting the alarm to away mode, when pressing arm away on the home-assistant panel, the alarm sends a medical alarm to the control station instead of arming away.

Integration is done via MQTT
mcu is esp8266

Arm home works without problems, here is my configuration:

ESPHome:
````
substitutions:
  systemName: "dscalarm"
  panelId: "dscalarm"

dsc_alarm_panel:
  id: "$panelId"
  usercodes: "0000:Master"
  accesscode: "0000"
  defaultpartition: "1"
  maxzones: "32"
  dscdebuglevel: "1"
  dscclockpin: "5"
  dscreadpin: "4"
  dscwritepin: "15"

external_components:
  - source: github://Dilbert66/esphome-components@main
    components: [dsc_alarm_panel,binary_sensor,text_sensor] 

esphome:
  name: $systemName
  platform: ESP8266
  board: nodemcuv2
  platformio_options:
    build_flags: 
        - "-Wl,-Map,output.map"  

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "$systemName"
    password: !secret wifi_password

logger:
  baud_rate: 115200
  level: DEBUG

mqtt:
  broker: !secret mqtt_broker
  port: 1883
  username: !secret mqtt_user
  password: !secret mqtt_pass
  topic_prefix: dscalarm

ota:
  - platform: esphome
    password: !secret ota_password

safe_mode:
    on_safe_mode:
      then:
       - switch.turn_off: connection_status_switch

status_led:
  pin:
    number: GPIO2
    inverted: no

binary_sensor:
  - platform: template
    id_code: z1
    name: "front_door_z1"
    device_class: door
    publish_initial_state: true 
  - platform: template
    id_code: z2
    name: "garage_door_z2"
    device_class: door
    publish_initial_state: true 
  - platform: template
    id_code: z3
    name: "office_window_z3"
    device_class: window
    publish_initial_state: true 
  - platform: template
    id_code: z4
    name: "motion_living_room_z4"
    device_class: motion
    publish_initial_state: true 
  - platform: template
    id_code: z5
    name: "main_front_window_z5"
    device_class: window
    publish_initial_state: true 
  - platform: template
    id_code: z6
    name: "dining_room_window_z6"
    device_class: window
    publish_initial_state: true 
  - platform: template
    id_code: z7
    name: "living_room_side_window_z7"
    device_class: window
    publish_initial_state: true 
  - platform: template
    id_code: z9
    name: "kitchen_door_z9"
    device_class: door
    publish_initial_state: true 
  - platform: template
    id_code: z10
    name: "main_bedroom_window_z10"
    device_class: window
    publish_initial_state: true 
  - platform: template
    id_code: z11
    name: "guest_bedroom_window_z11"
    publish_initial_state: true 
    device_class: window
  - platform: template
    id_code: z12
    name: "master_bedroom_bath_window_z12"
    device_class: window
    publish_initial_state: true 
  - platform: template
    id_code: z13
    name: "rooftop_door_z13"
    device_class: door
    publish_initial_state: true 
  - platform: template
    id_code: rdy_1
    name: "partition_1_ready"
    publish_initial_state: true 
  - platform: template
    id_code: arm_1
    name: "partition_1_armed"
    publish_initial_state: true 
  - platform: template
    id_code: rdy_2
    name: "partition_2_ready"
    publish_initial_state: true 
  - platform: template
    id_code: arm_2
    name: "partition_2_armed"
    publish_initial_state: true 
    #panel trouble status on/off
  - platform: template
    id_code: tr
    name: "alarm_trouble_status"
    device_class: problem
    publish_initial_state: true 

    #battery status ok/low
  - platform: template
    id_code: bat
    name: "alarm_battery_status"
    device_class: problem
    publish_initial_state: true 

    #AC power status ok/no power
  - platform: template
    id_code: ac
    name: "alarm_ac_status"
    device_class: plug
    publish_initial_state: true 

    #panic alarm on/off
  - platform: template
    id_code: panic
    name: "alarm_panic_status"
    device_class: safety
    publish_initial_state: true 
    #fire alarm on/off
  - platform: template
    id_code: fa_1
    device_class: safety
    publish_initial_state: true 
    name: "alarm_fire_partition_1_status"

    #fire alarm on/off
  - platform: template
    id_code: fa_2
    device_class: safety
    publish_initial_state: true 
    name: "alarm_fire_partition_2_status"

# alarm id code = al_ + partition number  
  - platform: template
    id_code: al_1  
    name: "partition_1_alarm_status"
    publish_initial_state: true        
    
  - platform: template
    id_code: al_2  
    name: "partition_2_alarm_status"  
    publish_initial_state: true

#relay PGM channels. Will show the state of the activate relay channel on/off
#uncomment the name: field if you wish to see it in home assistant
  - platform: template
    id_code: r1
    name: "pgm_1"
    publish_initial_state: true 
  - platform: template
    id_code: r2
    name: "pgm_2"
    publish_initial_state: true 
  - platform: template
    id_code: r3
    name: "pgm_3"
    publish_initial_state: true 
  - platform: template
    id_code: r4
    name: "pgm_4"
    publish_initial_state: true 
  

text_sensor:
  
    #general system status online/disconnected
  - platform: template
    id_code: system_status
    name: "alarm_system_status"
    icon: "mdi:shield"
    #battery level status for wireless channels, tamper , in alarm, etc for individual zones
  - platform: template
    id_code: zone_status
    name: "alarm_zone_status"
    icon: "mdi:shield"
    # partition status ie read/unavailable, in alarm, etc
# partition status ie ready/not ready, triggered, etc
# partition status id code = ps_ + partition number   
  - platform: template
    id_code: ps_1  
    name: "partition_1_status"
    icon: "mdi:shield"

  - platform: template
    id_code: ps_2  
    name: "partition_2_status" 
    icon: "mdi:shield"  

# more verbose message regarding the partition status. ie zones open, bypassed, etc
# partition message id code = msg_ + partition number    
  - platform: template
    id_code: msg_1  
    name: "partition_1_msg"
    icon: "mdi:alert-box"

  - platform: template
    id_code: msg_2  
    name: "partition_2_msg"
    icon: "mdi:alert-box"
# virtual lcd keypad line1 and line2 messages for each partition   
# partition line1 id code = ln1_ + partition number
# partition line2 id code = ln2_ + partition number
# partition 1
  - platform: template
    id_code: ln1_1  
    name: "line1_partition_1"
    icon: "mdi:alert-box"

  - platform: template
    id_code: ln2_1  
    name: "line2_partition_1"
    icon: "mdi:alert-box"  

# partition 2
  - platform: template
    id_code: ln1_2  
    name: "line1_partition_2"
    icon: "mdi:alert-box"

  - platform: template
    id_code: ln2_2  
    name: "line2_partition_2"
    icon: "mdi:alert-box" 

# generic event messages from panel
# panel event id code = evt
  - platform: template
    id_code: evt  
    name: "alarm_event"
    icon: "mdi:alert-box" 
  
# partition beeps
# beeps id code = bp_ + partition number
  - platform: template
    id_code: bp_1  
    name: "alarm_beeps"
    icon: "mdi:alert-box" 

  - platform: template
    id_code: bp_2  
    name: "partition_2 beeps"
    icon: "mdi:alert-box"   

 # panel trouble messages  
 # panel msg id code = tr_msg
  - platform: template
    id_code: tr_msg  
    name: "alarm_trouble_msg" 
    icon: "mdi:alert-box"

switch:
    #shows status of connection status to panel.  You can disconnect before upload using the switch.
  - platform: template
    name: "$systemName Connection"
    id: connection_status_switch
    lambda: |-
      return dsc.keybusConnected;
    icon: "mdi:shield-link-variant"
    turn_on_action:
      - switch.toggle: restart_switch
    turn_off_action:
      - lambda: |-
          disconnectKeybus();
  - platform: restart
    id: restart_switch
````

Alarm Panel in /homeassistant/configuration.yaml

````
mqtt:
  alarm_control_panel:
    - name: "dscalarm"
      state_topic: "dscalarm/sensor/partition_1_status/state"
      availability_topic: "dscalarm/status"
      command_topic: "dscalarm/alarm/set"
      code: 1111
      payload_disarm: '{"keys":"NNNN","partition":"1"}'
      payload_arm_home: '{"keys":"S","partition":"1"}'
      payload_arm_away: '{"keys":"A","partition":"1"}'
````

when pressing arm_home the alarm arms properly with home status however sending payload_arm_away results in a medical emergency being sent.
When pressing disarm the alarm disarms properly.

Thanks for your help!

jescarri, 2024-11-17:
I'm reading the code, it seems arm_away moved from A to W ? Is that the case?
https://github.com/Dilbert66/esphome-components/blob/main/components/dsc_alarm_panel/dscKeybusInterface.cpp#L411-L415

Thanks again for the help

Dilbert66, 2024-11-17:
Yes, A is auxiliary alarm, W is Away

jescarri, 2024-11-28:
Thanks this worked!. I'm closing the issue,

And again thank you for this software it has been working flawlessly since the early days!.

closed 2024-11-28 by jescarri