Dilbert66/esphome-components #10: The bot stopped responding after the last update

Allkorolev, 2024-12-23:
Today I updated the configuration on my device, the running bot stopped responding to requests. 

Changes that were made in the Yaml code:
- deleting mg_lib because an error appeared (Component mg_lib cannot be loaded via YAML (no CONFIG_SCHEMA).)
Before this there was a working version, it was last updated on October 25th, there was no longer a need for this, today I updated it and the bot stopped responding. There are no error messages in the esphome logs.

Dilbert66, 2024-12-23:
Add mg_lib to the external_components and recompile.  The mg_lib library and bot code were updated to a new version.

Allkorolev, 2024-12-24:
Got you, thanks for your work

closed 2024-12-24 by Allkorolev