Dilbert66/esphome-components #11: The bot does not respond to commands when it is mentioned

Allkorolev, 2024-12-26:
Hello. 
The idea is the following, I have several bots in my group that have the same command, for example /help
If you simply enter the /help command in the group, the bot will work normally, but if you enter /help@testbot with a mention of the bot, then the bot is silent.

The help command is written as follows:
`    - command: "/help"
      then: 
        - telegram.publish:
            message: !lambda return "text here"; 
            to: !lambda return x.chat_id;`

I understand that you can explicitly specify /help@testbot in the command and it will work, but the difficulty is that the bot can change, and sometimes it is not possible to rebuild the firmware.

Dilbert66, 2024-12-26:
Good idea, I'll implement it. I'll have something later today.. Easy change. You will need to add a "bot_name:" to to your yaml config so it can match against.

Allkorolev, 2024-12-26:
Thanks a lot
It would be nice to add switching between html and markdown markup and also disable disable_web_page_preview

Dilbert66, 2024-12-26:
Those options are already available. "parse_mode" and "disable_web_page_preview" under all actions.

Dilbert66, 2024-12-26:
Ok, i've pushed an update to dev.  On startup, the esp will fetch the current bot's name from the api. ie @botName unless you override it by adding a "bot_name:" entry to the yaml config.  The botname must start with @, ie  bot_name: @thisbotname.  Now will be able to access using /cmd xxxx or /cmd@botName xxxx

Allkorolev, 2024-12-27:
> Ok, i've pushed an update to dev. On startup, the esp will fetch the current bot's name from the api. ie @botName unless you override it by adding a "bot_name:" entry to the yaml config. The botname [...]

Did I understand correctly that now the bot's name will be determined automatically? Those. When loading a bot, will it be possible to access it directly in the group? 
I sent the command /help@testbot but the bot did not respond. 
Or do you need to specify something additional in YAML?
Can you give a short example?

Allkorolev, 2024-12-27:
> disable_web_page_preview

Can you advise on disable_web_page_preview
```
    - command: "/help,/h"
      then: 
        - telegram.publish:
            parse_mode: "html"
            disable_web_page_preview: true
            message: "text here"
            to: !lambda return x.chat_id;    
```
When compiling I get an error. The error message states that the set_disable_web_preview method does not exist and that you need set_disable_web_page_preview instead
With the second method I also get the same error.

Dilbert66, 2024-12-27:
From what I see, what you are trying to do doesnt work on telegram unless you do it from the bot channel.  It won't work in a group.  Telegram doesnt send the message to a bot user using /cmd@botname from what I see.  That's a telegram thing, not an issue with my code as it never reaches the bot code.

As to the disable_web_page_preview issue, that has been corrected.

Allkorolev, 2024-12-27:
> As to the disable_web_page_preview issue, that has been corrected.

I confirm disable_web_page_preview works

> Edit: Ok, I got it to work in the group. You have to add your bots to the administrators in the group so that this will work, then you will be able to do /cmd@BotId1 and /cmd@BotId2. And yes, the code will get the actual bot name from the channel. It will use the display name (not the username).

I added the bot to the channel administrators, I send the command /help@testbot - the command will not be executed, the bot does not respond.

One more moment, is it possible using 

`    - command: "/water"`

Run: 
`- callback: "/h2o"`

If so, how can this be done? When interacting with a bot, according to logic, the user can request the same data both through the command and through inline_keyboard

Dilbert66, 2024-12-27:
It works fine here. I have two bots in the same channel and can have them respond individually. Make sure when you send your cmds that you can see a parse args debug message in your esphome logs.  If you don't then it means telegram is not sending the message.
```
[telegram:158]:  parse args: cmd=/s,to=@DscBot,args=
[telegram:363]: TLS init - Before: freeheap: 170880,minheap: 87508,maxfree:116724

```

Dilbert66, 2024-12-27:
[image]

Dilbert66, 2024-12-27:
As to your callback question, I don't know. Never tried it. I'll have to look at the logic.

Dilbert66, 2024-12-27:
When you talk to your bot, make sure you use the correct case. Right now it's case sensitive and has to match the displayed name of the bot.

Allkorolev, 2024-12-28:
> As to your callback question, I don't know. I'll have to look at the logic. The callback code needs some reconfiguring anyhow so I'll have a better look.

I’ll try to explain the intended logic, there are 2 commands and 1 callback:

```
	- callback: "/call_water"
	  then:
		- telegram.publish:
		  message: "text here"
                  to: !lambda return x.chat_id;
		

    - command: "/start"
      then:
        - telegram.delete_message:
            message_id: !lambda return x.message_id;
            chat_id: !lambda return x.chat_id;  
        - telegram.publish:
            message: !lambda return "@"+ x.sender +", text here:";
            inline_keyboard: "[[{'text':'Water','callback_data':'/call_water'}]]"
            to: !lambda return x.chat_id; 
			
	- command: "/water"
	  then:

```
callback: "/call_water"
will contain a lot of dynamic data from sensors

command: "/start"
serves to call inline_keyboard which also contains a request for callback_water

The question is how to execute /call_water using the /water command 

This will help to get data both from inline_keyboard and using the command.

Dilbert66, 2024-12-28:
You cant' execute a callback  from an internal command.  The callback is a response to an inline request only.

Allkorolev, 2024-12-29:
I continue to fight the bot. 
Sending a message to the /h group
I receive a message from the bot as expected

In the logs at the time of contacting the bot: 
[12:15:23][D][telegram:163]: parse args: cmd=/w,to=,args=
I also noticed that after loading esp32 in the logs it was: [12:12:12][D][telegram:288]: Set bot name to testbot

If I send a command directly to the bot
/w@testbot
The bot does not process the command in the logs, I see the following:
[12:19:12][D][telegram:163]: parse args: cmd=/w,to=@testbot,args=

Dilbert66, 2024-12-29:
ah, found the issue. I had an @ in both my bot names so I always included the @ in the search while you did not for yours. I've corrected the query to omit the @

Allkorolev, 2024-12-29:
I checked it, it works but not quite as expected, I’ll give an example:

```
    - command: "/h,/help"
      then:
        - telegram.publish:
            parse_mode: "html"
            message: "/h or /help - call help menu"       
            to: !lambda return x.chat_id; 
```
In this case, the bot will respond with a message with the clickable commands /h and /help, when clicked, the command /h@testbot will be sent, where testbot is the bot’s login and, accordingly, the command will not be executed.

It would be very nice if the command was processed as /h@loginbot and not /h@namebot

Dilbert66, 2024-12-29:
I figured that might be an issue as I mentioned that earlier. Anyhow, I've modified the code to use the username instead of the display name for the lookup.

Allkorolev, 2024-12-29:
I checked, now everything works fine. Thank you very much

closed 2024-12-29 by Allkorolev