Dilbert66/esphome-components #12: MG_EV_ERROR

bds89, 2025-01-01:
Hi, thanks for the components. I use your telegram bot on esp32, from time to time I get MG_EV_ERROR server in the log which is fixed only by rebooting the board. Telegram bot does not work with this error. Any ideas?

Dilbert66, 2025-01-01:
You need to post some logs so we can see what is happening.

bds89, 2025-01-03:
Thanks, the problem is solved. I changed the power supply esp32 and wifi settings in my router, some of this helped.

Allkorolev, 2025-01-08:
Today I updated the library, a similar problem. 
Before this, everything worked fine, there were no restrictions on the network, the power supply was tested

```
19:29:20 [D] [telegram:589] Connecting to telegram...
19:29:20 [E] [telegram:503] MG_EV_ERROR occured. Retrying in 15 seconds.
19:29:22 [D] [telegram:608] Stack high water mark: 10276
19:29:37 [D] [telegram:589] Connecting to telegram...
19:29:38 [D] [telegram:391] TLS init - Before: freeheap: 100000,minheap: 29448,maxfree:44020
19:29:38 [D] [telegram:400] TLS init - After: freeheap: 56180,minheap: 29448,maxfree:44020
19:31:28[D] [telegram:589] Connecting to telegram...
```

Allkorolev, 2025-01-08:
> Thanks, the problem is solved. I changed the power supply esp32 and wifi settings in my router, some of this helped.

Tell me in more detail what changes you made?

Dilbert66, 2025-01-08:
You need to post more logs before the ev_error so we can see why your app is retrying. Just posting the ev_error message is not helfpful.   The previous changes done were to prevent a crash when an ev_error occored, and now instead retries gracefully after 15 seconds.  They would not cause a connection issue.  You have something going on with your network it seems.

Dilbert66, 2025-01-08:
Also looks like you are getting tight on ram.  You can set "stack_size: 8" under the telegram: module to set stack size to 8k instead of the preset default of 16 to give you an additional 8k.  A larger stack than normal is needed in telegram if you find that your bot logic starts getting complex especially if you have a lot of if: statements.  The ssl logic needs at least 40-50k of heap ram to connect.  I don't see any oom errors so I don't think that's the issue in your case.  Looks to more network related.

Dilbert66, 2025-01-08:
Also be aware that the bot will reconnect every 1 to 2 minutes to the api as it is using long polling.

Dilbert66, 2025-01-08:
I realized that esphome logs will not show the mongoose lib connection error unless you view it from the serial port. I've added a specific esphome log print so now you will see the connection error reason.

Dilbert66, 2025-01-08:
You see those because you are monitoring from the serial port. If you were monitoring via the esphome dashboard, you would not have seen those.  Now, they will show.   Those errors indicate a network issue.  Dns resolving or other. Not related to the telegram code.

Dilbert66, 2025-01-08:
I actually removed the connect timeout. Looks like there is some delay in your connection to the api so would cause the connection to fail.   It should help with your connect erors now.  Good you posted the logs so we can see what the actual issue is.

fedsanchez, 2025-01-08:
Please excuse my ignorance, but would the changes you made effect my recompiling even if I am using a local file repository?

Dilbert66, 2025-01-08:
No, they would not.

fedsanchez, 2025-01-09:
Got it.  I changed to your repository and it works fine.  But the error now shows:

[21:12:27][E][vista_alarm:1262][cmdQueueTask]: Data timeout. Is the panel connected?
[21:12:30]10c0f  1 mongoose.c:434:mg_error        8 -1 DNS timeout
[21:12:30][E][telegram:502]: MG_EV_ERROR 8 -1 DNS timeout. Retrying in 15 seconds.
[21:12:56]17217  1 mongoose.c:434:mg_error        10 -1 DNS timeout
[21:12:56][E][telegram:502]: MG_EV_ERROR 10 -1 DNS timeout. Retrying in 15 seconds.
[21:13:14]1b877  1 mongoose.c:434:mg_error        11 -1 DNS timeout
[21:13:14][E][telegram:502]: MG_EV_ERROR 11 -1 DNS timeout. Retrying in 15 seconds.
[21:13:32]1fed7  1 mongoose.c:434:mg_error        12 -1 DNS timeout
[21:13:32][E][telegram:502]: MG_EV_ERROR 12 -1 DNS timeout. Retrying in 15 seconds.

Is this my connection issue as well?

Dilbert66, 2025-01-09:
Are you using a manual ip setup? What is your dns1 entry?

fedsanchez, 2025-01-09:
manual_ip:
    static_ip: $static_wifi_ip
    gateway: 10.0.0.1
    subnet: 255.255.255.0
    dns1: 192.168.0.1             

I moved a wifi repeater to 10 feet from the ESP32.  Now I get these error:

[21:30:04][E][web_server:2159]: MG_EV_ERROR 4 54 socket error.
[21:34:24][E][web_server:2159]: MG_EV_ERROR 7 52 socket error.
[21:37:02][E][web_server:2159]: MG_EV_ERROR 6 54 socket error.
[21:41:08][E][web_server:2159]: MG_EV_ERROR 11 55 socket error.

Dilbert66, 2025-01-09:
If your gateway is 10.0.0.1 shouldnt your dns be 10.0.0.1? You should not have a gateway and dns on different subnets..  Your network connection is not right.  Why don't you let the esp get it's own ip and dns instead of using a manual entry?

fedsanchez, 2025-01-09:
I don't remember why I did the manual setup.  Somewhere I read or was told to do it, as it would be better.  I would not have done it otherwise.  As far as the dns, you told me to use 192.168.0.1

I got rid of the manual setting, and there is nothing to replace it with, correct? Just:

wifi:
  ssid: $ssid
  password: $ssid_pw

Now I am getting:

[22:52:21]13960  1 mongoose.c:434:mg_error        8 54 socket error
[22:52:21][E][web_server:2159]: MG_EV_ERROR 8 54 socket error.
[22:52:21]13969  1 mongoose.c:434:mg_error        7 53 socket error
[22:52:21][E][web_server:2159]: MG_EV_ERROR 7 53 socket error.

fedsanchez, 2025-01-09:
I remember now why I  used a manual ip.  So that I could locate the web panel easier, and have 2 going at the same time - one that is active and in use, and one for testing new changes.

I changed back to manual, but made the dns the same as the gateway, correct?  Like this:

  manual_ip:
    static_ip: $static_wifi_ip
    gateway: 10.0.0.1
    subnet: 255.255.255.0
    dns1: 10.0.0.1         

I still get the errors:

[23:06:16]578a0  1 mongoose.c:434:mg_error        4 53 socket error
[23:06:16][E][web_server:2159]: MG_EV_ERROR 4 53 socket error.

fedsanchez, 2025-01-09:
This seems to be holding steady:

  manual_ip:
    static_ip: $static_wifi_ip
    gateway: 10.0.0.1
    subnet: 255.255.255.0
    dns1: 8.8.8.8               # google dns
    dns2: 10.0.0.1            

Thanks!

btw, is there a way to get the error to be posted less frequently:

[00:05:26][E][vista_alarm:1262][cmdQueueTask]: Data timeout. Is the panel connected?

Dilbert66, 2025-01-09:
HA and Esphome use mdns to locate their devices. Using the devicename and appending .local.  If you get the error that often, you have issues on your network so you need to address them. That's why they are there. I will remove the duplicate message though as it's not needed.

closed 2025-01-12 by Dilbert66