Dilbert66/esphome-components #13: Today's update broke the integration

adlertag1, 2025-01-04:
The updated old binary_sensor/text_sensor to work same as alarm templates broke the integration. The api does't send any data. Reverting back to the previous works.

Dilbert66, 2025-01-04:
Are you using dev or master branch? What components do you load?

adlertag1, 2025-01-04:
Hi! I'm using master branch. The binary_sensor and text_sensor components are no longer functional with the latest update.  I would also like to take this opportunity to thank you for the excellent work you’ve done on this code. I’ve been using it for a long time, and it has always worked flawlessly until now. Your work is greatly appreciated!

Dilbert66, 2025-01-04:
I've reverted the old commit so all should be good now. Tks for pointing it out. Glad you are finding the project useful!

adlertag1, 2025-01-04:
Thank you so much for the quick response and for reverting the commit! I really appreciate your prompt action and the incredible effort you put into this project.
I hope the root cause of the issue becomes clear soon, and if there's anything I can do to assist in troubleshooting, please let me know. 
Thanks again for your dedication and support!

Dilbert66, 2025-01-04:
The root of the issue was me as I thought I was in dev and commited an update to master instead :)

closed 2025-01-04 by adlertag1