Dilbert66/esphome-components #15: Unable to compile dev branch for ESP8266

XD36JD2015, 2025-01-23:
I am unable to compile and install the source files from the dev branch for my ESP8266 device. 

[logs_alarm-system-shop_compile.txt](https://github.com/user-attachments/files/18514523/logs_alarm-system-shop_compile.txt)

[alarm-system-shop.txt](https://github.com/user-attachments/files/18514541/alarm-system-shop.txt)

Dilbert66, 2025-01-23:
This should be fixed now on latest update.

XD36JD2015, 2025-01-24:
Hi, thanks for fixing that so quickly.  Unfortunately, I have different compile errors now.

[logs_alarm-system-shop_run.txt](https://github.com/user-attachments/files/18528807/logs_alarm-system-shop_run.txt)

Dilbert66, 2025-01-24:
I removed the offending taskyied call. You should be good now.

XD36JD2015, 2025-01-24:
Compile was successful this time, though there are 3 warnings that occured during the compile.  Thanks again!

[logs_alarm-system-shop_run(1).txt](https://github.com/user-attachments/files/18528966/logs_alarm-system-shop_run.1.txt)

closed 2025-01-31 by XD36JD2015