Dilbert66/esphome-components #16: Zones in alarm functionality

duanemck, 2025-02-18:
Hi, thanks for the amazing library, I was using the DSCKeybus library manually with an Arduino sketch but this solution is much more elegant from an HA integration point of view.

One thing I'm missing is a way to expose which zone has triggered the alarm. The DSCKeybus library does expose this and it works in a very similar way to the `dsc.openZonesChanged` feature:

```
dsc.alarmZonesStatusChanged
//and
dsc.alarmZonesChanged[zoneGroup]
```
I was looking at the code and considering doing a PR but I'm not sure about the best way to expose this in HA. I see 2 options:

1. Expose an additional `binary_sensor` for each zone that is "true" when that zone triggered the alarm. The downside is many additional sensors that aren't used very often until a partition is triggered.
2. Find some way to track the state for each sensor that is not binary. i.e. `Open/Closed/Alarmed|Triggered` but that is quite a substantial change from the current `binary_sensor`s. 

Would you have any preference? Or other ideas?

Dilbert66, 2025-02-18:
The code already supports alarm zones.  I suggest you try the "dev" branch. It is more advanced.

duanemck, 2025-03-17:
Thank you, I missed your edits. I'll give it a try this week.

duanemck, 2025-04-06:
Thanks, have updated to dev and see the new sensors. Haven't been able to test properly as triggering the alarm is a pain, but I'll get it setup so long anyway.

closed 2025-04-06 by duanemck