Dilbert66/esphome-components #20: Zone state read with great delay after upgrade/restart

weirded, 2025-04-01:
When I update the firmware and/or restart the ESP, it takes a long time (~ several minutes) for zones that are open to correctly show "open" state again in Home Assistant. I ran an earlier version (before it converted to a `custom_component`) previously that didn't seem to suffer from this issue. 

Anything I can configure differently to make the state update more immediately?

fedsanchez, 2025-04-01:
I assume that you're talking about RF sensors, not wired. 

This is likely due to a change that I had requested, and was implemented. Previously, when the system status was "ready" all the sensors were assumed to be closed. Now, they only show closed when it receives the RF signal stating as such. If you open and close the sensor, you will see it is reported as open, and then will be reported as closed.

I think this is better, because it allows for unused sensors to be monitored without influencing the alarm. It also shows you the information that was received, not an assumption. 

When I see no status on a sensor, I know it hasn't changed from when I booted the ESP.

Dilbert66, 2025-04-01:
I"m going to assume you are talking about the DSC system.  The panel only sends full zone status updates every few minutes so on an esp reboot  statuses won't be available until then. The extremely old component code was probably using something I was trying out to store statuses to temp non volatile storage which was then re-applied to restart but this proved problematic from what I remember. I don't have a better solution at this time and that code was removed in later iterations.   What you can do though to force a panel refresh after an esp restart is to enter this code sequence at the panel or the virtual keypad:

*8xxxx# 

where xxxx is your installer code.  This will put the system go into installer mode then exit , forcing a refresh.

fedsanchez, 2025-04-01:
Sorry, my bad.

fedsanchez, 2025-04-01:
Sorry, my bad.

Dilbert66, 2025-04-02:
You can do it in home assistant but it would need to be done right after the esphome device starts up.
You can also do it in esphome using on_boot and a delay. You then call the alarm_keypress_partition service to send the cmds.  You can use a substitution to hide the code view in source or send it directly
```
substitutions:
  installcode: !secret install_code
```
```
esphome:
   on_boot:
     - priority: -100 
       then:
         - delay: 30s
         - lambda: |-
             $panelId->alarm_keypress_partition("*8${installcode}#",1);
```

weirded, 2025-04-02:
Thanks. Circling back to this - this seems to work, but only if I hard code the install_code into the lambda call... Not sure the String substitution works in the lambda?

Dilbert66, 2025-04-02:
Yes, substitutions work perfectly in a lambda. You have to use the format ${nameofvar}

weirded, 2025-04-02:
Got it working... Thank you for the help and thanks for maintaining this project! :)

closed 2025-04-02 by weirded