Dilbert66/esphome-components #21: ESP seems to drop from the network when partition is armed/disarm

duanemck, 2025-04-06:
When I arm or disarm a partition (via HA or via the alarm itself), the ESP8266 seems to drop from the network for a bit. 
Looking at the logs, it actions the command (so the alarm does actually arm/disarm) but then I see it drop from the network and come back a short while later. 

On HA it shows as `unavailable` while this is happening.

Here's the logs from the ESPHome add-on. I've set the log level to VERBOSE but I'm not really seeing any errors
```
[22:31:34][V][text_sensor:013]: 'Beams Status': Received new state armed_away
[22:31:34][D][text_sensor:064]: 'Beams Status': Sending state 'armed_away'
WARNING dsc-alarm @ 192.168.1.13: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for dsc-alarm @ 192.168.1.13
WARNING Disconnected from API
INFO Successfully connected to dsc-alarm @ 192.168.1.13 in 0.006s
INFO Successful handshake with dsc-alarm @ 192.168.1.13 in 2.650s
```

I'm using the `dev` branch and I have disabled the web keypad

duanemck, 2025-04-06:
Changed log level to `VERY_VERBOSE` and saw the following at around the time I armed the alarm:

```
[22:57:30][VV][scheduler:225]: Running interval 'update' with interval=0 last_execution=907 (now=39491)
[22:57:30][D][api:103]: Accepted 192.168.1.11
[22:57:30][VV][api.socket:648]: 192.168.1.11: noise_handshakestate_new_by_id failed: NO_MEMORY
[22:57:30][W][api.connection:122]: : Socket operation failed: HANDSHAKESTATE_SETUP_FAILED errno=11
[22:57:30][VV][scheduler:225]: Running interval 'update' with interval=0 last_execution=907 (now=39518)
[22:57:30][V][api:116]: Removing connection to 192.168.1.11
[22:57:30][VV][scheduler:225]: Running interval 'update' with interval=0 last_execution=907 (now=39533)
```

Dilbert66, 2025-04-06:
I recommend you use an ESP32.  The esp8266 is too limited in resources to run the dev version properly as it looks like you are running out of ram.

duanemck, 2025-07-21:
Managed to migrate to ESP32 this weekend and can confirm that the issue is solved, thanks for the assistance.

closed 2025-07-21 by duanemck