Dilbert66/esphome-components #25: Zone Status for zones above 32 appears in the incorrect partition

DeepPurple77, 2025-06-08:
Hi @Dilbert66,

I took your advice and switched to the Dev branch, after some fiddling I'm back and up running.

I was having this issue with the main branch and it hasn't resolved with the dev branch so reporting it - I think it's an off by 1 again (hopefully).

It's a little tricky to explain but if I go to Bypass Zones (It's * then * and then I can use the arrow to go through each zone), only zones below 32 are appearing for parition 1 (there should be some above 32). The same is true when a zone above 32 is open, it's not showing even though the system isn't ready.

In partition 2, when going into the Bypass Zone section all zones above 32 are actually the parition 1 zones and I don't see parition 2 zones. I've verified that this isn't the case on a pysical keypad and the zones are correctly set in the panel.

I tested in both Home Assistant and the new on-chip Alarm Panel (via the browser) and both show the same issue.

Here's a log of going through the Zone Bypasses for parition 2 (Zone 33, Zone 37 and Zone 38 should be in partition 1):
[21:19:07][D][text_sensor:064]: 'line1 Partition 2 (ln1_2)': Sending state 'System is Ready'
[21:19:07][D][text_sensor:064]: 'line2 Partition 2 (ln2_2)': Sending state 'Ready to Arm <>'
[21:19:07][D][text_sensor:064]: 'Partition 2 Msg (msg_2)': Sending state '01 Ready'
[21:19:07][D][text_sensor:064]: 'Partition 2 Status (ps_2)': Sending state 'ready'
[21:19:08][I][Panel :966]: [2025-06-08 21:19] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[21:19:09][I][Panel :966]: [2025-06-08 21:19] C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 (34)
[21:19:09][I][dscalarm:2064]: status 01, last status 01,selection 01,partition=1,skip=0,force=1
[21:19:09][I][dscalarm:2064]: status 01, last status 01,selection 01,partition=2,skip=0,force=1
[21:19:09][I][dscalarm:2064]: status 01, last status 01,selection 01,partition=3,skip=0,force=1
[21:19:14][I][Panel :966]: [2025-06-08 21:19] E6: E6 00 0B 00 F1 00 00 00 00 00 00 00 00 00 00 00 (34)
[21:19:22][I][dscalarm:805]: Writing keys: * to partition 2, partition disabled: 0 , partition locked: 0
[21:19:22][I][dscalarm:2064]: status 01, last status 01,selection 01,partition=2,skip=0,force=1
[21:19:23][I][Module:966]: [2025-06-08 21:19] 05: FF 01 FF 28 FF FF FF FF FF FF 01 00 00 00 00 00 (34)
[21:19:23][I][Panel :966]: [2025-06-08 21:19] 63: 63 00 00 00 00 00 00 63 00 00 00 00 00 00 00 00 (58)
[21:19:23][D][sensor:094]: 'Uptime s': Sending state 3253.26611 s with 0 decimals of accuracy
[21:19:23][I][Panel :966]: [2025-06-08 21:19] E6: E6 00 18 02 00 00 00 00 00 00 00 00 00 00 00 00 (74)
[21:19:23][I][Panel :966]: [2025-06-08 21:19] 0F: 0F 00 80 9E 00 00 00 00 00 2D 00 00 00 00 00 00 (74)
[21:19:23][I][Panel :966]: [2025-06-08 21:19] E6: E6 00 21 80 9E 00 00 00 00 80 A5 00 00 00 00 00 (82)
[21:19:23][I][Panel :966]: [2025-06-08 21:19] 05: 05 00 81 01 80 9E 81 01 00 C7 01 00 00 00 00 00 (74)
[21:19:23][I][dscalarm:2064]: status 9E, last status 01,selection 01,partition=2,skip=0,force=0
[21:19:23][D][text_sensor:064]: 'line1 Partition 2 (ln1_2)': Sending state 'Press (*) for <>'
[21:19:23][D][text_sensor:064]: 'line2 Partition 2 (ln2_2)': Sending state 'Zone Bypass'
[21:19:23][D][text_sensor:064]: 'Partition 2 Msg (msg_2)': Sending state '158 Enter * code'
[21:19:23][D][binary_sensor:036]: 'Partition 2 Ready (rdy_2)': Sending state OFF
[21:19:23][D][text_sensor:064]: 'Partition 2 Status (ps_2)': Sending state 'not_ready'
[21:19:24][I][Panel :966]: [2025-06-08 21:19] E6: E6 00 17 04 00 00 00 00 00 01 00 00 00 00 00 00 (74)
[21:19:24][I][Panel :966]: [2025-06-08 21:19] E6: E6 00 18 04 00 00 00 00 00 02 00 00 00 00 00 00 (74)
[21:19:26][I][dscalarm:805]: Writing keys: * to partition 2, partition disabled: 0 , partition locked: 0
[21:19:26][I][Module:966]: [2025-06-08 21:19] 05: FF 01 FF 28 FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[21:19:26][I][Module:966]: [2025-06-08 21:19] 05: FF 01 FF 05 FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[21:19:26][I][Panel :966]: [2025-06-08 21:19] 63: 63 00 08 00 00 00 00 6B 00 00 00 00 00 00 00 00 (58)
[21:19:26][I][Panel :966]: [2025-06-08 21:19] E6: E6 00 18 02 08 00 00 00 00 08 00 00 00 00 00 00 (74)
[21:19:26][I][Panel :966]: [2025-06-08 21:19] A5: A5 00 25 99 15 4E 66 00 2C 00 00 00 00 00 00 00 (66)
[21:19:27][I][Panel :966]: [2025-06-08 21:19] 0F: 0F 00 80 A0 00 00 00 00 00 2F 00 00 00 00 00 00 (74)
[21:19:27][I][Panel :966]: [2025-06-08 21:19] E6: E6 00 21 80 A0 00 00 00 00 80 A7 00 00 00 00 00 (82)
[21:19:27][I][Panel :966]: [2025-06-08 21:19] 05: 05 00 81 01 80 A0 81 01 00 C7 01 00 00 00 00 00 (74)
[21:19:27][I][dscalarm:2064]: status A0, last status 9E,selection 00,partition=2,skip=0,force=1
[21:19:27][D][text_sensor:064]: 'line1 Partition 2 (ln1_2)': Sending state 'Zone bypass <>'
[21:19:27][D][text_sensor:064]: 'line2 Partition 2 (ln2_2)': Sending state '01 Garage Door'
[21:19:27][I][dscalarm:2064]: status A0, last status A0,selection 01,partition=2,skip=0,force=1
[21:19:27][D][text_sensor:064]: 'Partition 2 Msg (msg_2)': Sending state '160 Zone bypass'
[21:19:27][I][Panel :966]: [2025-06-08 21:19] EB: EB 00 02 25 19 15 4C 02 66 00 F4 00 00 00 00 00 (82)
[21:19:27][D][text_sensor:064]: 'event (evt)': Sending state '[2025.06.08 21:19] Zone bypass program, Partition 2 status is not_ready'
[21:19:27][I][dscalarm:2064]: status A0, last status A0,selection 01,partition=2,skip=0,force=1
[21:19:29][I][Panel :966]: [2025-06-08 21:19] 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00 (67)
[21:19:29][I][dscalarm:2064]: status A0, last status A0,selection 01,partition=2,skip=0,force=1
[21:19:29][I][Module:966]: [2025-06-08 21:19] 11: FF 01 CF CC FC F3 CF FF FF 03 00 00 00 00 00 00 (67)
[21:19:30][I][dscalarm:805]: Writing keys: > to partition 2, partition disabled: 0 , partition locked: 0
[21:19:30][I][dscalarm:2064]: status A0, last status A0,selection 05,partition=2,skip=0,force=1
[21:19:30][D][text_sensor:064]: 'line2 Partition 2 (ln2_2)': Sending state '05 Kitchen Door'
[21:19:30][I][Module:966]: [2025-06-08 21:19] 05: FF 01 FF 87 FF FF FF FF FF FF 01 00 00 00 00 00 (67)
[21:19:33][D][sensor:094]: 'WiFi Signal dB': Sending state -71.00000 dBm with 0 decimals of accuracy
[21:19:36][I][dscalarm:805]: Writing keys: > to partition 2, partition disabled: 0 , partition locked: 0
[21:19:36][I][dscalarm:2064]: status A0, last status A0,selection 11,partition=2,skip=0,force=1
[21:19:36][D][text_sensor:064]: 'line2 Partition 2 (ln2_2)': Sending state '17 BY 2 PIR'
[21:19:37][I][Module:966]: [2025-06-08 21:19] 05: FF 01 FF 87 FF FF FF FF FF FF 01 00 00 00 00 00 (67)
[21:19:39][I][dscalarm:805]: Writing keys: > to partition 2, partition disabled: 0 , partition locked: 0
[... 48 lines cut ...]
[21:19:59][I][Module:966]: [2025-06-08 21:19] 11: FF 01 CF CC FC F3 CF FF FF 03 00 00 00 00 00 00 (67)
[21:20:00][I][dscalarm:805]: Writing keys: > to partition 2, partition disabled: 0 , partition locked: 0
[21:20:00][I][dscalarm:2064]: status A0, last status A0,selection 00,partition=2,skip=0,force=1
[21:20:00][D][text_sensor:064]: 'line2 Partition 2 (ln2_2)': Sending state '01 Garage Door'
[21:20:01][I][Module:966]: [2025-06-08 21:20] 05: FF 01 FF 87 FF FF FF FF FF FF 01 00 00 00 00 00 (67)
[21:20:04][I][dscalarm:805]: Writing keys: # to partition 2, partition disabled: 0 , partition locked: 0
[21:20:04][I][dscalarm:2064]: status A0, last status A0,selection FF,partition=2,skip=0,force=1
[21:20:04][D][text_sensor:064]: 'line2 Partition 2 (ln2_2)': Sending state ' '
[21:20:04][I][Module:966]: [2025-06-08 21:20] 05: FF 01 FF 2D FF FF FF FF FF FF 01 00 00 00 00 00 (67)
[21:20:04][I][Panel :966]: [2025-06-08 21:20] 63: 63 00 00 00 00 00 00 63 00 00 00 00 00 00 00 00 (58)
[21:20:04][I][dscalarm:2064]: status A0, last status A0,selection FF,partition=2,skip=0,force=1
[21:20:04][I][Panel :966]: [2025-06-08 21:20] E6: E6 00 18 02 00 00 00 00 00 00 00 00 00 00 00 00 (74)
[21:20:04][I][dscalarm:2064]: status A0, last status A0,selection FF,partition=2,skip=0,force=1
[21:20:05][I][Panel :966]: [2025-06-08 21:20] 05: 05 00 81 01 81 01 81 01 00 C7 01 00 00 00 00 00 (74)
[21:20:05][I][dscalarm:2064]: status 01, last status A0,selection FF,partition=2,skip=0,force=0
[21:20:05][D][text_sensor:064]: 'line1 Partition 2 (ln1_2)': Sending state 'System is Ready'
[21:20:05][D][text_sensor:064]: 'line2 Partition 2 (ln2_2)': Sending state 'Ready to Arm <>'
[21:20:05][D][text_sensor:064]: 'Partition 2 Msg (msg_2)': Sending state '01 Ready'
[21:20:05][D][binary_sensor:036]: 'Partition 2 Ready (rdy_2)': Sending state ON
[21:20:05][D][text_sensor:064]: 'Partition 2 Status (ps_2)': Sending state 'ready'
[21:20:14][I][Panel :966]: [2025-06-08 21:20] E6: E6 00 2C 40 00 00 00 00 52 00 00 00 00 00 00 00 (66)

Apologies for the lenghty debug log, I'm not sure what exactly may be relevant.
Regards
Dale

Dilbert66, 2025-06-08:
Definitively a code issue. Unfortunately I have an 1832 here which does not support zones above 32 so I could not properly test that part of it.  Can you do me a favor and run a cmd while logging?  I need to see some more info regarding the enabled zones processes.  Basically all I need you to do is get into program mode then exit.  Ie do a *8<installercode> then exit with ##.  

Also, please indicate which zones belong to partition 1 and those that belong to partition 2.  
Don't forget to remove your installer code from the logs. It will show up under "writing xxxx"

DeepPurple77, 2025-06-09:
Hi Alan,

Thanks again for looking at that so quickly. I tested with your updates and it's a little better in that Partition 1 zones are no longer appearing on Partition 2 but I don't see any zones above 32 on either Partition 1 or Partition 2. I also noticed today that zone 24 from partion 3 is appearing in partition 1 (but zone 25 isn't - weird)!

Here are the zones by partition:
**Partition 1**
2 3 4 6 7 8 26 33 37 38
**Partition 2**
1 5 17 18 19 20 21 22 23 27 28 35 36 39 41 42 (41 and 42 are on emulated zones on the ESP32)
**Partition 3**
24 25

Debug Log below
Thank you!
Dale

[10:58:57][I][dscalarm:805]: Writing keys: * to partition 1, partition disabled: 0 , partition locked: 0
[10:58:57][I][dscalarm:2063]: status 01, last status 01,selection 01,partition=1,skip=0,force=1
[10:58:57][I][Module:966]: [2025-06-09 10:58] 05: FF 01 28 FF FF FF FF FF FF FF 01 00 00 00 00 00 (67)
[10:58:57][I][Panel :966]: [2025-06-09 10:58] 5D: 5D 00 00 00 00 00 00 5D 00 00 00 00 00 00 00 00 (58)
[10:58:57][I][Panel :966]: [2025-06-09 10:58] E6: E6 00 18 01 00 00 00 00 00 FF 00 00 00 00 00 00 (74)
[10:58:57][I][Panel :966]: [2025-06-09 10:58] 0A: 0A 00 80 9E 00 00 00 00 00 28 00 00 00 00 00 00 (74)
[10:58:57][I][Panel :966]: [2025-06-09 10:58] E6: E6 00 20 80 9E 00 00 00 00 80 A4 00 00 00 00 00 (82)
[10:58:57][I][Panel :966]: [2025-06-09 10:58] 05: 05 00 80 9E 81 02 81 01 00 C7 01 00 00 00 00 00 (74)
[10:58:57][I][dscalarm:2063]: status 9E, last status 01,selection 01,partition=1,skip=0,force=0
[10:58:57][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'Press (*) for <>'
[10:58:57][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state 'Zone Bypass'
[10:58:57][D][text_sensor:064]: 'Partition 1 Msg (msg_1)': Sending state '9e Enter * code'
[10:58:57][D][binary_sensor:036]: 'Partition 1 Ready (rdy_1)': Sending state OFF
[10:58:57][D][text_sensor:064]: 'Partition 1 Status (ps_1)': Sending state 'not_ready'
[10:58:57][I][dscalarm:805]: Writing keys: 8 to partition 1, partition disabled: 0 , partition locked: 0
[10:58:57][I][Module:966]: [2025-06-09 10:58] 05: FF 01 22 FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[10:58:58][I][Panel :966]: [2025-06-09 10:58] 05: 05 00 80 B7 81 02 81 01 00 C7 01 00 00 00 00 00 (74)
[10:58:58][I][dscalarm:2063]: status B7, last status 9E,selection 00,partition=1,skip=0,force=0
[10:58:58][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'Enter'
[10:58:58][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state 'installer code'
[10:58:58][D][text_sensor:064]: 'Partition 1 Msg (msg_1)': Sending state 'b7 Installer code'
[10:59:00][I][dscalarm:805]: Writing keys: <redacted> to partition 1, partition disabled: 0 , partition locked: 0
[10:59:00][I][Module:966]: [2025-06-09 10:59] 05: FF 01 <redacted> FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[10:59:00][I][dscalarm:805]: Writing keys: <redacted> to partition 1, partition disabled: 0 , partition locked: 0
[10:59:00][I][Module:966]: [2025-06-09 10:59] 05: FF 01 <redacted> FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[10:59:01][I][dscalarm:805]: Writing keys: <redacted> to partition 1, partition disabled: 0 , partition locked: 0
[10:59:01][I][Module:966]: [2025-06-09 10:59] 05: FF 01 <redacted> FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[10:59:02][I][dscalarm:805]: Writing keys: <redacted> to partition 1, partition disabled: 0 , partition locked: 0
[10:59:02][I][Module:966]: [2025-06-09 10:59] 05: FF 01 <redacted> FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[10:59:02][I][Panel :966]: [2025-06-09 10:59] 64: 64 00 0C 70 00 00 00 00 00 00 00 00 00 00 00 00 (26)
[10:59:02][D][text_sensor:064]: 'beeps (bp_1)': Sending state '6'
[10:59:02][D][text_sensor:064]: 'partition 2 beeps (bp_2)': Sending state '6'
[10:59:02][D][text_sensor:064]: 'partition 3 beeps (bp_3)': Sending state '6'
[10:59:02][I][dscalarm:2063]: status B7, last status B7,selection 00,partition=1,skip=0,force=0
[10:59:02][I][dscalarm:2063]: status 02, last status 02,selection 01,partition=2,skip=0,force=0
[10:59:02][I][dscalarm:2063]: status 01, last status 01,selection 01,partition=3,skip=0,force=0
[10:59:03][I][Panel :966]: [2025-06-09 10:59] 16: 16 00 0E 42 A6 0C 00 00 00 00 00 00 00 00 00 00 (42)
[10:59:03][I][Panel :966]: [2025-06-09 10:59] 5D: 5D 00 20 00 00 00 00 7D 00 00 00 00 00 00 00 00 (58)
[10:59:03][I][Panel :966]: [2025-06-09 10:59] E6: E6 00 18 01 20 00 00 00 00 1F 00 00 00 00 00 00 (74)
[10:59:03][I][Panel :966]: [2025-06-09 10:59] 0A: 0A 00 82 E4 00 00 00 00 00 70 00 00 00 00 00 00 (74)
[10:59:03][I][Panel :966]: [2025-06-09 10:59] E6: E6 00 20 82 E4 00 00 00 00 80 EC 00 00 00 00 00 (82)
[10:59:03][I][Panel :966]: [2025-06-09 10:59] 05: 05 00 82 E4 81 C7 81 C7 00 C7 01 00 00 00 00 00 (74)
[10:59:03][I][dscalarm:2063]: status E4, last status B7,selection 00,partition=1,skip=0,force=0
[10:59:03][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'Section:'
[10:59:03][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state '(3 digits)'
[10:59:03][D][text_sensor:064]: 'Partition 1 Msg (msg_1)': Sending state 'e4 Installer menu'
[10:59:03][I][Panel :966]: [2025-06-09 10:59] A5: A5 00 25 19 2A E9 AD FF A2 00 00 00 00 00 00 00 (66)
[10:59:03][I][Panel :966]: [2025-06-09 10:59] EB: EB 00 00 25 19 2A E8 01 AD FF E8 00 00 00 00 00 (82)
[10:59:03][D][text_sensor:064]: 'event (evt)': Sending state '[2025.06.09 10:58] Enter inst prog, Partition status is not_ready'
[... cut ...]
[10:59:22][I][Panel :966]: [2025-06-09 10:59] 34: 34 00 85 02 81 02 00 3E 00 00 00 00 00 00 00 00 (58)
[10:59:22][I][Panel :966]: [2025-06-09 10:59] 3E: 3E 00 85 02 81 02 00 48 00 00 00 00 00 00 00 00 (58)
[10:59:22][I][Panel :966]: [2025-06-09 10:59] 27: 27 00 85 02 81 02 14 45 00 00 00 00 00 00 00 00 (58)
[10:59:34][I][Panel :966]: [2025-06-09 10:59] E6: E6 00 0B 00 F1 00 00 00 00 00 00 00 00 00 00 00 (34)
[10:59:35][I][Panel :966]: [2025-06-09 10:59] 27: 27 00 85 02 81 02 10 41 00 00 00 00 00 00 00 00 (58)

DeepPurple77, 2025-06-09:
Hi Alain,

You are an absolute star, thank you for trying to figure this out!

I think I can see the problem, it looks like Zones higher than 32 are getting mapped to 1 partition more than they should - e.g. Zone 35 is showing as enabled on Partition 3 whereas it should be on Partition 2. It's not quite clear because 33 is showing as being on Partition 1 but isn't showing up anywhere as mentioned earlier.

Anyway, thanks again - debug is below!

[16:31:16][I][dscalarm:805]: Writing keys: * to partition 1, partition disabled: 0 , partition locked: 0
[16:31:16][I][dscalarm:2067]: status 01, last status 01,selection 01,partition=1,skip=0,force=1
[16:31:16][I][Module:966]: [2025-06-09 16:31] 05: FF 01 28 FF FF FF FF FF FF FF 01 00 00 00 00 00 (34)
[16:31:16][I][Panel :966]: [2025-06-09 16:31] 5D: 5D 00 00 00 00 00 00 5D 00 00 00 00 00 00 00 00 (58)
[16:31:16][I][Panel :966]: [2025-06-09 16:31] E6: E6 00 18 01 00 00 00 00 00 FF 00 00 00 00 00 00 (74)
[16:31:16][I][Panel :966]: [2025-06-09 16:31] 0A: 0A 00 80 9E 00 00 00 00 00 28 00 00 00 00 00 00 (74)
[16:31:16][I][Panel :966]: [2025-06-09 16:31] E6: E6 00 20 80 9E 00 00 00 00 80 A4 00 00 00 00 00 (82)
[16:31:16][I][dscalarm:805]: Writing keys: 8 to partition 1, partition disabled: 0 , partition locked: 0
[16:31:16][I][dscalarm:2067]: status 01, last status 01,selection 01,partition=1,skip=0,force=1
[16:31:16][I][Panel :966]: [2025-06-09 16:31] 05: 05 00 80 9E 81 01 81 01 00 C7 01 00 00 00 00 00 (74)
[16:31:16][I][dscalarm:2067]: status 9E, last status 01,selection 01,partition=1,skip=0,force=0
[16:31:16][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'Press (*) for <>'
[16:31:16][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state 'Zone Bypass'
[16:31:16][D][text_sensor:064]: 'Partition 1 Msg (msg_1)': Sending state '9e Enter * code'
[16:31:16][D][binary_sensor:036]: 'Partition 1 Ready (rdy_1)': Sending state OFF
[16:31:16][D][text_sensor:064]: 'Partition 1 Status (ps_1)': Sending state 'not_ready'
[16:31:16][I][Module:966]: [2025-06-09 16:31] 05: FF 01 22 FF FF FF FF FF FF FF 01 00 00 00 00 00 (74)
[16:31:17][I][Panel :966]: [2025-06-09 16:31] 05: 05 00 80 B7 81 01 81 01 00 C7 01 00 00 00 00 00 (74)
[16:31:17][I][dscalarm:2067]: status B7, last status 9E,selection 01,partition=1,skip=0,force=0
[16:31:17][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'Enter'
[16:31:17][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state 'installer code'
[16:31:17][D][text_sensor:064]: 'Partition 1 Msg (msg_1)': Sending state 'b7 Installer code'
[16:31:17][I][Panel :966]: [2025-06-09 16:31] A5: A5 00 25 19 30 80 00 00 93 00 00 00 00 00 00 00 (66)
[16:31:20][I][dscalarm:805]: Writing keys: XX to partition 1, partition disabled: 0 , partition locked: 0
[16:31:20][I][Module:966]: [2025-06-09 16:31] 05: FF 01 XX FF FF FF FF FF FF FF 01 00 00 00 00 00 (66)
[16:31:20][I][dscalarm:805]: Writing keys: XX to partition 1, partition disabled: 0 , partition locked: 0
[16:31:21][I][Module:966]: [2025-06-09 16:31] 05: FF 01 XX FF FF FF FF FF FF FF 01 00 00 00 00 00 (66)
[16:31:21][I][dscalarm:805]: Writing keys: XX to partition 1, partition disabled: 0 , partition locked: 0
[16:31:21][I][Module:966]: [2025-06-09 16:31] 05: FF 01 XX FF FF FF FF FF FF FF 01 00 00 00 00 00 (66)
[16:31:21][I][dscalarm:805]: Writing keys: XX to partition 1, partition disabled: 0 , partition locked: 0
[16:31:21][I][Module:966]: [2025-06-09 16:31] 05: FF 01 XX FF FF FF FF FF FF FF 01 00 00 00 00 00 (66)
[16:31:21][I][Panel :966]: [2025-06-09 16:31] 64: 64 00 0C 70 00 00 00 00 00 00 00 00 00 00 00 00 (26)
[16:31:21][D][text_sensor:064]: 'beeps (bp_1)': Sending state '6'
[16:31:21][D][text_sensor:064]: 'partition 2 beeps (bp_2)': Sending state '6'
[16:31:21][D][text_sensor:064]: 'partition 3 beeps (bp_3)': Sending state '6'
[16:31:22][I][dscalarm:2067]: status B7, last status B7,selection 01,partition=1,skip=0,force=0
[16:31:22][I][dscalarm:2067]: status 01, last status 01,selection 01,partition=2,skip=0,force=0
[16:31:22][I][dscalarm:2067]: status 01, last status 01,selection 01,partition=3,skip=0,force=0
[16:31:22][I][Panel :966]: [2025-06-09 16:31] 16: 16 00 0E 42 A6 0C 00 00 00 00 00 00 00 00 00 00 (42)
[16:31:22][I][Panel :966]: [2025-06-09 16:31] 5D: 5D 00 20 00 00 00 00 7D 00 00 00 00 00 00 00 00 (58)
[16:31:22][I][Panel :966]: [2025-06-09 16:31] E6: E6 00 18 01 20 00 00 00 00 1F 00 00 00 00 00 00 (74)
[16:31:22][I][Panel :966]: [2025-06-09 16:31] 0A: 0A 00 82 E4 00 00 00 00 00 70 00 00 00 00 00 00 (74)
[16:31:22][I][Panel :966]: [2025-06-09 16:31] E6: E6 00 20 82 E4 00 00 00 00 80 EC 00 00 00 00 00 (82)
[16:31:22][I][Panel :966]: [2025-06-09 16:31] 05: 05 00 82 E4 81 C7 81 C7 00 C7 01 00 00 00 00 00 (74)
[16:31:22][I][dscalarm:2067]: status E4, last status B7,selection 01,partition=1,skip=0,force=0
[16:31:22][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'Section:'
[16:31:22][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state '(3 digits)'
[16:31:22][D][text_sensor:064]: 'Partition 1 Msg (msg_1)': Sending state 'e4 Installer menu'
[16:31:22][I][Panel :966]: [2025-06-09 16:31] A5: A5 00 25 19 30 81 AD FF 40 00 00 00 00 00 00 00 (66)
[16:31:23][I][Panel :966]: [2025-06-09 16:31] EB: EB 00 00 25 19 30 80 01 AD FF 86 00 00 00 00 00 (82)
[16:31:23][D][text_sensor:064]: 'event (evt)': Sending state '[2025.06.09 16:32] Enter inst prog, Partition status is not_ready'
[... cut ...]
[16:31:29][I][Panel :966]: [2025-06-09 16:31] 2D: 2D 00 85 01 81 01 00 35 00 00 00 00 00 00 00 00 (58)
[16:31:29][I][Panel :966]: [2025-06-09 16:31] 34: 34 00 85 01 81 01 00 3C 00 00 00 00 00 00 00 00 (58)
[16:31:29][I][Panel :966]: [2025-06-09 16:31] 3E: 3E 00 85 01 81 01 00 46 00 00 00 00 00 00 00 00 (58)
[16:31:29][I][Panel :966]: [2025-06-09 16:31] 27: 27 00 85 01 81 01 00 2F 00 00 00 00 00 00 00 00 (58)
[16:31:29][I][Panel :966]: [2025-06-09 16:31] E6: E6 00 09 00 EF 00 00 00 00 00 00 00 00 00 00 00 (34)

Dilbert66, 2025-06-09:
Perfect, I think I got it now.  Another issue with the partition fetch routine. I've pushed a new update to dev.

DeepPurple77, 2025-06-09:
You are a genius! I don't know how you figure these issues out! The Zones are now working perfectly but there's a new problem !

My emulated zones (41 and 42) are "stuck" open.  I can no longer get them to close at all - they show open on both Home Assistant and on the DSC Panel itself. This was a snippet from trying to open Zone 41, the switch doesn't take effect even though HA shows the zone is closed the panel says differently.

[20:03:31][D][switch:012]: 'Open Driveway Gate Zone' Turning ON.
[20:03:31][I][Module:968]: [2025-06-09 20:03] 05: FF 01 FF FF FF FF FF BF FF FF 01 00 00 00 00 00 (74)
[20:03:31][I][Panel :968]: [2025-06-09 20:03] E6: E6 00 0A FF FF FF FF FF FF 01 00 00 00 00 00 00 (66)
[20:03:31][I][Module:968]: [2025-06-09 20:03] E6: FF 01 FF 5F 5F FF FF 8F FF 01 00 00 00 00 00 00 (66)
[20:03:31][I][Panel :968]: [2025-06-09 20:03] E6: E6 00 0B 03 F4 00 00 00 00 00 00 00 00 00 00 00 (34)
[20:03:32][I][Module:968]: [2025-06-09 20:03] 05: FF 01 FF FF FF FF FF 7F FF FF 01 00 00 00 00 00 (34)
[20:03:32][I][Panel :968]: [2025-06-09 20:03] E6: E6 00 08 FF FF FF FF FF FF 01 00 00 00 00 00 00 (66)
[20:03:32][I][Module:968]: [2025-06-09 20:03] E6: FF 01 FF FF FF 11 15 F8 FF 01 00 00 00 00 00 00 (66)
[20:03:32][I][Panel :968]: [2025-06-09 20:03] E6: E6 00 09 20 0F 00 00 00 00 00 00 00 00 00 00 00 (34)
[20:03:32][D][text_sensor:064]: 'zone status (zs)': Sending state 'OP:41,OP:42,OP:38'

Dilbert66, 2025-06-09:
Can you post the log when you send the turn off cmd to the zone?

DeepPurple77, 2025-06-09:
Hmmm, I now understand that it's not about setting the zoneFault to On or Off but actually getting the state of the zone. Without state the switch immediately turns off but the zoneFault is left On.

My switch configuration in ESPHOME for Zone 41 is below and has been working perfectly up until the latest fixes. I even went back to the previous version with a local install and the zone works correctly with the below configuation.

**Previous working configuration with state**
binary_sensor:   
  - platform: dsc_alarm_panel
    id_code: z41
    id: z41
    name: "Driveway Gate (virtual)"
    device_class: garage_door
    publish_initial_state: true

switch:
  - platform: template
    name: "Open Driveway Gate Zone"
    id: switch_drivewayGate
    lambda: |-
      if (id(z41).state) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      - lambda: |-
         dsc.setZoneFault(41,1);
    turn_off_action:
      - lambda: |-
         dsc.setZoneFault(41,0);

I removed the lambda for state and added in optimistic so it doesn't care about the state and now it's working fine, so I think it isn't able to get state for z41 with the latest changes.

**Switch configuration without state working with latest changes:**
switch:
  - platform: template
    name: "Open Driveway Gate Zone"
    id: switch_drivewayGate
    optimistic: True
    turn_on_action:
      - lambda: |-
         dsc.setZoneFault(41,1);
    turn_off_action:
      - lambda: |-
         dsc.setZoneFault(41,0);

**Debug log in any event!**
[21:39:18][D][switch:012]: 'Open Driveway Gate Zone' Turning ON.
[21:39:18][D][switch:055]: 'Open Driveway Gate Zone': Sending state ON
[21:39:18][I][Module:968]: [2025-06-09 21:39] 05: FF 01 FF FF FF FF FF BF FF FF 01 00 00 00 00 00 (67)
[21:39:18][I][Panel :968]: [2025-06-09 21:39] E6: E6 00 0A FF FF FF FF FF FF 01 00 00 00 00 00 00 (66)
[21:39:18][I][Module:968]: [2025-06-09 21:39] E6: FF 01 FF 57 00 FF FF CF FF 01 00 00 00 00 00 00 (66)
[21:39:18][I][Panel :968]: [2025-06-09 21:39] E6: E6 00 0B 01 F2 00 00 00 00 00 00 00 00 00 00 00 (34)
[21:39:18][D][text_sensor:064]: 'zone status (zs)': Sending state 'OP:41'
[21:39:24][D][switch:016]: 'Open Driveway Gate Zone' Turning OFF.
[21:39:24][D][switch:055]: 'Open Driveway Gate Zone': Sending state OFF
[21:39:24][I][Module:968]: [2025-06-09 21:39] 05: FF 01 FF FF FF FF FF BF FF FF 01 00 00 00 00 00 (34)
[21:39:24][I][Panel :968]: [2025-06-09 21:39] E6: E6 00 0A FF FF FF FF FF FF 01 00 00 00 00 00 00 (66)
[21:39:24][I][Module:968]: [2025-06-09 21:39] E6: FF 01 FF 55 57 FF FF 6F FF 01 00 00 00 00 00 00 (66)
[21:39:24][I][Panel :968]: [2025-06-09 21:39] E6: E6 00 0B 00 F1 00 00 00 00 00 00 00 00 00 00 00 (34)
[21:39:24][D][text_sensor:064]: 'zone status (zs)': Sending state ''

Thanks again!
Dale

Dilbert66, 2025-06-09:
That's normal.  That also affects zones below 32.  The reason is that the panel only sends zone status updates every few minutes and you won't see the actual zone status until that happens.  Going into installer mode then exiting, forces an update from the panel.  That's why I had you doing that. Nothing I can do  to speed that up.

DeepPurple77, 2025-06-09:
Ok perfect, I thought that may be the answer! Thanks so much again, you’re an amazing maintainer! I’ll go ahead and close the issue…