Dilbert66/esphome-components #26: Latest espHome breaks compilation

rwoldberg, 2025-07-16:
I updated to the latest version of ESPHome and the module no longer compiles, I get this error:

ERROR Unable to import component binary_sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 200, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 999, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/data/external_components/c9cfb228/components/binary_sensor/__init__.py", line 64, in <module>
    from esphome.cpp_helpers import setup_entity
ImportError: cannot import name 'setup_entity' from 'esphome.cpp_helpers' (/esphome/esphome/cpp_helpers.py)
Failed config

binary_sensor: [source /config/esphome/vista20p-bridge.yaml:180]
  
  Component not found: binary_sensor.

Dilbert66, 2025-07-16:
Yes, I'm aware and working on updates.

rwoldberg, 2025-07-16:
Ok, thank you.

Dilbert66, 2025-07-17:
I've pushed a complete new version. Unfortunately you will need to update your yaml config. Please see the example yaml from the project page.  Please note also the added api section config line "custom_services: true".

rwoldberg, 2025-07-17:
Working perfectly.  Thanks.

DeepPurple77, 2025-07-17:
@Dilbert66 Not sure if I should hijack this issue or start a new one but here I am...

Using Esphome 2025.7.0 and 2025.7.1 with the new release in the Dev branch, I get weird behaviour. The build compiles now but once the ESP32 is booted, I get logs and it doesn't communicate with HA. I added the custom_services:true and didn't see any other changes in your dscalarm.yaml file - so not sure what is going on here.

I had to go back 2025.6.3 and the commit from 10 June to get my ESP32 back online and working. Surprisingly, OTA worked as expected.

Dilbert66, 2025-07-17:
Interesting, it works perfectly here on my end. Has no issue with two way communications with HA. Don't know what to say.  Double check the yaml again just to make sure.  The custom_service line is to enable HA to send cmds to the esp but is not needed just for sending data back to HA. It enables the key send service.

DeepPurple77, 2025-07-17:
Thanks, I'll go to a clean YAML config and see how that works.

For what it's worth, even the Web Keypad didn't work - almost like some sort of deadlock.

Dilbert66, 2025-07-17:
Looks like it might be rebooting. Only know for sure would be to see the serial logs from the device with a pc directly connected to it.  The wifi logs won't show the reboot reason.

GreatHolyCow, 2025-07-18:
There was quite a few changes to Yaml, id_code is now id and platform: template is now platform: dsc_alarm_panel. With update and latest stable esphome i have finally gotten mine back online