Dilbert66/esphome-components #27: AlarmBinarySensor::set_trigger_on_initial_state undefined on ESPHome 2025.7.2

carldebilly, 2025-07-19:
I'm getting those errors on ESP Home 2025.7.2

Using
```yml
external_components:
  - source: github://Dilbert66/esphome-components@main
    components: [dsc_alarm_panel]
```

Building with ESPHome 2025.7.2 fails :
```
/config/esphome/alarmXXXX.yaml: In function 'void setup()':
error: 'class esphome::alarm_panel::AlarmBinarySensor' has no member named 'set_trigger_on_initial_state'; did you mean 'set_publish_initial_state'?
```

Full excerpt (same error repeated for every dsc_alarm_panel binary sensor):
```
/config/esphome/alarmXXXX.yaml:109:10: error: 'class esphome::alarm_panel::AlarmBinarySensor' has no member named 'set_trigger_on_initial_state'; did you mean 'set_publish_initial_state'?
/config/esphome/alarmXXXX.yaml:126:10: error: 'class esphome::alarm_panel::AlarmBinarySensor' has no member named 'set_trigger_on_initial_state'; did you mean 'set_publish_initial_state'?
/config/esphome/alarmXXXX.yaml:145:12: error: 'class esphome::alarm_panel::AlarmBinarySensor' has no member named 'set_trigger_on_initial_state'; did you mean 'set_publish_initial_state'?
```

`set_trigger_on_initial_state()` was removed in ESPHome 2025.5 and replaced by `set_publish_initial_state()`.
Could you update `dsc_alarm_panel` to call the new method (or add a compatibility shim) so it builds on current ESPHome versions?

Dilbert66, 2025-07-19:
The code works fine and was updated.  set_trigger_initial_state() is correct for version 2025.7.x.  I'm not sure what is going on with your setup but it's not picking up the right esphome code.  They obsoleted set_publish_inital state.

carldebilly, 2025-07-22:
After a clean-up of the compilation files, it works. Thanks!

closed 2025-07-22 by carldebilly