Dilbert66/esphome-components #28: ESPHome 2025.7.x and ESP32 reboots continously.

DeepPurple77, 2025-07-19:
Hi Alain,

Picking up from where I previously posted. You were right that the ESP32 is rebooting continously and causing no connectivity back to Home Assistant, although it does seem to stay up occasionally for a short period of time.

I performed a serial debug and it seems to be complaining about the following, this line is repeated hundreds of times before the ESP eventually reboots: 
**gptimer: gptimer_start(396): timer is not ready for a new start**
It's always gptimer_start(396) - I'm not sure if 396 means something?

A complete restart / reboot loop is below, I've cut the majority of the gptimer_start lines to save a bit of space and indicated so with **\<cut\>**

Going back to ESPHome 2025.6.3 and reverting to the last commit you made in June restores functionality and everything works perfectly. I'm not sure how but OTA works even while the ESP32 seems to be "stuck".

Regards
Dale

```
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
E (599) esp_core_dump_flash: No corf"յ⸮ partition found!
E (599) esp_core_dump_flash: No core dump partition found!
[0;32m[I][logger:169]: Log initialized[0m
[0;35m[C][safe_mode:082]: There have been 3 suspected unsuccessful boot attempts[0m
[0;36m[D][esp32.preferences:142]: Writing 1 items: 0 cached, 1 written, 0 failed[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 41[0m
[0;36m[D][binary_sensor:026]: 'Driveway Gate (virtual)': New state is OFF[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 42[0m
[0;36m[D][binary_sensor:026]: 'Patio PIR (virtual)': New state is OFF[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 43[0m
[0;36m[D][binary_sensor:026]: 'TEST ZONE (virtual)': New state is OFF[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 1[0m
[0;36m[D][binary_sensor:026]: 'Garage Door': New state is OFF[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 2[0m
[0;36m[D][binary_sensor:026]: 'Study': New state is OFF[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 3[0m
[0;36m[D][binary_sensor:026]: 'Passage': New state is OFF[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 4[0m
[0;36m[D][binary_sensor:026]: 'Kitchen': New state is OFF[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 5[0m
[0;36m[D][binary_sensor:026]: 'Kitchen Door': New state is OFF[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 6[0m
[0;36m[D][binary_sensor:026]: 'Front Door': New state is OFF[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 7[0m
[0;36m[D][binary_sensor:026]: 'Patio Door': New state is OFF[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 8[0m
[0;36m[D][binary_sensor:026]: 'Playroom PIR': New state is OFF[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 17[0m
[0;36m[D][binary_sensor:026]: 'Backyard Door PIR': New state is OFF[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 18[0m
[0;36m[D][binary_sensor:026]: 'Driveway PIR': New state is OFF[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 19[0m
[0;36m[D][binary_sensor:026]: 'Front Garden PIR': New state is OFF[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 20[0m
[0;36m[D][binary_sensor:026]: 'Garage PIR': New state is OFF[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 21[0m
[0;36m[D][binary_sensor:026]: 'Front Garden Door': New state is OFF[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 22[0m
[0;36m[D][binary_sensor:026]: 'Shed Door': New state is OFF[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 23[0m
[0;36m[D][binary_sensor:026]: 'Garage Entrance to House': New state is OFF[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 24[0m
[0;36m[D][binary_sensor:026]: 'Domestic Door': New state is OFF[0m
[0;36m[D][dscalarm:4724]: CreatefromOjb: added zone 25[0m
[... 145 lines cut ...]
A6      : 0x3ffbdcf8  A7      : 0x00000001  A8      : 0x801115a8  A9      : 0x3ffc068c  
A10     : 0x00000001  A11     : 0x00000021  A12     : 0x00000014  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffa  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core  0 was running in ISR context:
EPC1    : 0x40112a5b  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace: 0x401115a5:0x3ffc06ac |<-CORRUPTED
Core  1 register dump:
PC      : 0x4000921a  PS      : 0x00060d35  A0      : 0x80007d16  A1      : 0x3ffc0d1c  
A2      : 0x00800000  A3      : 0x30000000  A4      : 0x00000000  A5      : 0x3f43895c  
A6      : 0x00400000  A7      : 0x00000016  A8      : 0x3ff40000  A9      : 0x00000033  
A10     : 0x00800000  A11     : 0x3ff4001c  A12     : 0x00000016  A13     : 0x3ffc0e9c  
A14     : 0x3ffbdd84  A15     : 0xffffffff  SAR     : 0x00000004  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4008ac6c  LEND    : 0x4008ac77  LCOUNT  : 0xffffffff  

Backtrace: 0x40009217:0x3ffc0d1c |<-CORRUPTED
ELF file SHA256: 01694f60a

E (11423) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
E (11432) esp_core_dump_elf: Elf write init failed!
E (11436) esp_core_dump_common: Core dump write failed with error=-1
Rebooting...

```

DeepPurple77, 2025-07-19:
It's very strange. I'll try to get a different ESP32 at some stage, for now I'll stay on 2025.6.3 and the June commit which is working like a dream! Thanks @Dilbert66!

Dilbert66, 2025-07-19:
I'll post an update shortly with some changes and see if it fixes it.

Dilbert66, 2025-07-19:
Ok, i've posted an updated that makes some small changes on the timer handling. Works fine here so hopefully that will address your issue.

DeepPurple77, 2025-07-19:
I’ve said it before and I’ll say it again, Alain, you are a genius! Problem resolved!

Thank you sir!

Dilbert66, 2025-07-19:
Awesome!

closed 2025-07-19 by Dilbert66