Dilbert66/esphome-components #34: Compilation failure with ESHome 2025.8.0 (with PR)

DeepPurple77, 2025-08-20:
Hi @Dilbert66,

Updated to ESPHome 2025.8.0 and wasn't able to compile a new firmware.

I used ChatGPT to figure out the problem and I've created a PR to fix the issue (https://github.com/Dilbert66/esphome-components/pull/33).

I hope this is the correct procedure!

Regards

```src/esphome/components/dsc_alarm_panel/dscAlarm.cpp: In member function 'virtual void esphome::alarm_panel::DSCkeybushome::setup()':
src/esphome/components/dsc_alarm_panel/dscAlarm.cpp:175:37: error: no match for 'operator=' (operand types are 'std::vector<esphome::binary_sensor::BinarySensor*>' and 'const esphome::StaticVector<esphome::binary_sensor::BinarySensor*, 45>')
  175 |       bMap = App.get_binary_sensors();
      |                                     ^
In file included from /data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/vector:72,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/functional:64,
                 from /data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/HardwareSerial.h:49,
                 from /data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:202,
                 from src/esphome/core/macros.h:7,
                 from src/esphome/core/defines.h:2,
                 from src/esphome/components/dsc_alarm_panel/dscAlarm.h:6,
                 from src/esphome/components/dsc_alarm_panel/dscAlarm.cpp:3:
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/vector.tcc:210:5: note: candidate: 'constexpr std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(const std::vector<_Tp, _Alloc>&) [with _Tp = esphome::binary_sensor::BinarySensor*; _Alloc = std::allocator<esphome::binary_sensor::BinarySensor*>]'
  210 |     vector<_Tp, _Alloc>::
      |     ^~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/vector.tcc:211:42: note:   no known conversion for argument 1 from 'const esphome::StaticVector<esphome::binary_sensor::BinarySensor*, 45>' to 'const std::vector<esphome::binary_sensor::BinarySensor*>&'
  211 |     operator=(const vector<_Tp, _Alloc>& __x)
      |               ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
In file included from /data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/vector:66:
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/stl_vector.h:766:7: note: candidate: 'constexpr std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(std::vector<_Tp, _Alloc>&&) [with _Tp = esphome::binary_sensor::BinarySensor*; _Alloc = std::allocator<esphome::binary_sensor::BinarySensor*>]'
  766 |       operator=(vector&& __x) noexcept(_Alloc_traits::_S_nothrow_move())
      |       ^~~~~~~~
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/stl_vector.h:766:26: note:   no known conversion for argument 1 from 'const esphome::StaticVector<esphome::binary_sensor::BinarySensor*, 45>' to 'std::vector<esphome::binary_sensor::BinarySensor*>&&'
  766 |       operator=(vector&& __x) noexcept(_Alloc_traits::_S_nothrow_move())
      |                 ~~~~~~~~~^~~
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/stl_vector.h:788:7: note: candidate: 'constexpr std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(std::initializer_list<_Tp>) [with _Tp = esphome::binary_sensor::BinarySensor*; _Alloc = std::allocator<esphome::binary_sensor::BinarySensor*>]'
  788 |       operator=(initializer_list<value_type> __l)
      |       ^~~~~~~~
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/stl_vector.h:788:46: note:   no known conversion for argument 1 from 'const esphome::StaticVector<esphome::binary_sensor::BinarySensor*, 45>' to 'std::initializer_list<esphome::binary_sensor::BinarySensor*>'
  788 |       operator=(initializer_list<value_type> __l)
      |                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
src/esphome/components/dsc_alarm_panel/dscAlarm.cpp:176:35: error: no match for 'operator=' (operand types are 'std::vector<esphome::text_sensor::TextSensor*>' and 'const esphome::StaticVector<esphome::text_sensor::TextSensor*, 23>')
  176 |       tMap = App.get_text_sensors();
      |                                   ^
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/vector.tcc:210:5: note: candidate: 'constexpr std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(const std::vector<_Tp, _Alloc>&) [with _Tp = esphome::text_sensor::TextSensor*; _Alloc = std::allocator<esphome::text_sensor::TextSensor*>]'
  210 |     vector<_Tp, _Alloc>::
      |     ^~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/vector.tcc:211:42: note:   no known conversion for argument 1 from 'const esphome::StaticVector<esphome::text_sensor::TextSensor*, 23>' to 'const std::vector<esphome::text_sensor::TextSensor*>&'
  211 |     operator=(const vector<_Tp, _Alloc>& __x)
      |               ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/stl_vector.h:766:7: note: candidate: 'constexpr std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(std::vector<_Tp, _Alloc>&&) [with _Tp = esphome::text_sensor::TextSensor*; _Alloc = std::allocator<esphome::text_sensor::TextSensor*>]'
  766 |       operator=(vector&& __x) noexcept(_Alloc_traits::_S_nothrow_move())
      |       ^~~~~~~~
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/stl_vector.h:766:26: note:   no known conversion for argument 1 from 'const esphome::StaticVector<esphome::text_sensor::TextSensor*, 23>' to 'std::vector<esphome::text_sensor::TextSensor*>&&'
  766 |       operator=(vector&& __x) noexcept(_Alloc_traits::_S_nothrow_move())
      |                 ~~~~~~~~~^~~
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/stl_vector.h:788:7: note: candidate: 'constexpr std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(std::initializer_list<_Tp>) [with _Tp = esphome::text_sensor::TextSensor*; _Alloc = std::allocator<esphome::text_sensor::TextSensor*>]'
  788 |       operator=(initializer_list<value_type> __l)
      |       ^~~~~~~~
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/stl_vector.h:788:46: note:   no known conversion for argument 1 from 'const esphome::StaticVector<esphome::text_sensor::TextSensor*, 23>' to 'std::initializer_list<esphome::text_sensor::TextSensor*>'
  788 |       operator=(initializer_list<value_type> __l)
      |                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
*** [.pioenvs/dscalarminterface/src/esphome/components/dsc_alarm_panel/dscAlarm.cpp.o] Error 1```

Dilbert66, 2025-08-20:
Tks Im already working on updating all projects.

DeepPurple77, 2025-08-20:
Issue has been resolved with commit https://github.com/Dilbert66/esphome-components/commit/1b2098635a43d17d3fb1ad1c37d673ca088a3430.

Thanks as always @Dilbert66!

closed 2025-08-20 by DeepPurple77