Dilbert66/esphome-components #37: API connection fails

yakirg, 2025-09-02:
After updating to latest, and updating yaml, I am getting the same as https://github.com/Dilbert66/esphome-components/issues/26
My config:
```
#for documentation see project at https://github.com/Dilbert66/esphome-VistaECP
substitutions:
  name: "vistaalarm" #unique network name, system name
  friendlyName: "Vista Alarm"  #used as the friendly name of your application in HomeAssistant
  panelId: "VistaAlarm"         #used as the service variable name.
  accesscode: !secret access_code 
  
vista_alarm_panel:
  id: $panelId
  accesscode: $accesscode #Only comes into effect if needed for arming and quickarm is not set
  maxzones: "48" #maximum amount of zones that your panel supports
  maxpartitions: "1" #maximum amount of partitions that your panel supports

  # Enroll your RF serial devices here. Format: serial#:loop#:zone#  Each record is comma separated.
  # For most devices loop1 is used such as 5800pir, other devices such as 5816 will use loop2.  Please refer to your
  # RF device programming (*56 program) to see what loop and zones are assigned to your RF devices. 
  # You can refer to the pdf link below for more details on loop numbers 
  # https://advancedsecurityllc.com/wp-content/uploads/5800%20Wireless%20Device%20List.pdf
  rfseriallookup: "0019994:2:66,0818433:4:22,0123456:1:55,0399512:1:17" # serial1:loop#:zone1,serial2:loop#:zone2
 
  defaultpartition: "1" #set to your main partition
  vistadebuglevel: "3" #component debug level for messages

  #assign a new virtual keypad address to each active partition using programs *190 - *196
  #and enter it below.  For unused partitions, use 0 as the keypad address.
  keypadaddr1: "19" #partition 1 virtual keyapd
  keypadaddr2: "0" #partition 2 virtual keypad. set to 0 to disable
  keypadaddr3: "0" #partition 3 virtual keypad. set to 0 to disable
 
  #AUI address from program field *189 to use for zone status queries (open/close and bypass). 
  #Ensure it is not assigned to a real keypad or to Total Connect 2.0. For those panels, you can select auiaddr: 5 or 6.
  #As a final option, you can assign it to the same address as your existing AUI display address of 1
  #Note: Older vista20 panels only have addresses 1 and 2 while newer will have 1,2,5 and 6. For Vista128,Vista250 commercial panels, ensure the address used is setup as an AUI keypad in program #93, device programming
  
  auiaddr: "5" #use auiaddr: 5 or 6 if you already have address 2 assigned to Total Connect or another keypad. Defaults to 0 - disabled
  ##esp32
  #rxpin: "22" #GPIO pin to use for data receive (yellow line) 
  #txpin: "21" #GPIO pin to use for data transmit (green line)
  #monitorpin: "18" #GPIO pin to use for monitoring module traffic such as RF or Expanders . Set to -1 to disable
  
  ##esp8266
  rxpin: "12" #GPIO pin to use for data receive (yellow line) 
  txpin: "14" #GPIO pin to use for data transmit (green line)
  monitorpin: "13" #GPIO pin to use for monitoring module traffic such as RF or Expanders . Set to -1 to disable
  
  #note for the vista128 and vista250, the expanderaddr and relayaddr addresses must be 0
  # module addresses:
  # 07 4229 zone expander  zones 9-16
  # 08 4229 zone expander zones 17-24
  # 09 4229 zone expander zones 25-32
  # 10 4229 zone expander zones 33-40
  # 11 4229 zone expander zones 41 48
  # 12 4204 relay module  
  # 13 4204 relay module
  # 14 4204 relay module
  # 15 4204 relay module
  
  expanderaddr1: "0" # 1st zone expander emulator (4229) address to use . Set to 0 to disable. 
  expanderaddr2: "0" # 2nd expander emulator address to use . Set to 0 to disable. 
  
  relayaddr1: "0" # relay module emulation (4204) addresses. Set to 0 to disable
  relayaddr2: "0"
  relayaddr3: "0"
  relayaddr4: "0"
  
  ttl: "30000"  # time  to live  in ms for zone/fire status before expiring;
  quickarm: "false"
[... 292 lines cut ...]
  update_interval: 30s

sensor:
  - platform: debug
    free:
      name: "Heap Free"
    block:
      name: "Heap Max Block"
    loop_time:
      name: "Loop Time"   
    #fragmentation: #esp8266 only
     # name: "Heap Fragmentation"      
```
Log is: 
```
INFO ESPHome 2025.8.2
INFO Reading configuration /config/esphome/alarmgateway.yaml...
INFO Detected timezone 'America/New_York'
INFO Starting log output from 192.168.88.74 using esphome API
INFO Successfully resolved vistaalarm @ 192.168.88.74 in 0.000s
WARNING Can't connect to ESPHome API for vistaalarm @ 192.168.88.74: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.88.74', port=6053))]: [Errno 111] Connect call failed ('192.168.88.74', 6053) (SocketAPIError)
INFO Trying to connect to vistaalarm @ 192.168.88.74 in the background
INFO Successfully resolved vistaalarm @ 192.168.88.74 in 0.000s
```
What am I missing?

forth29skywars, 2025-09-02:
Give this file a try.[fix.zip](https://github.com/user-attachments/files/22103565/fix.zip)

Dilbert66, 2025-09-03:
Yes, there is an issue with the esp8266 code that I'm looking into.  This error is not present on the esp32

Dilbert66, 2025-09-03:
I've pushed a new update that should fix the crashes on start.  Give it another try.

closed 2025-09-05 by Dilbert66